Hexxeh/rpi-firmware #252: Set the I2C Address of MPU6050

DevynCJohnson, 2020-12-15:
The MPU6050 (or at least the models that I purchase and use) allow the address to be changed by setting the "ADO" pin high, thus making the address 0x69. I do this with my MPU6050 modules since the DS1307 (the RTC) also uses the I2C address 0x68. However, the MPU6050 overlay does not have a parameter for setting the address.

Here is a link to a picture of the exact model of MPU6050 that I use - [https://pinouts.net/wp-content/uploads/2019/10/mpu-6050-pinouts.jpg](https://pinouts.net/wp-content/uploads/2019/10/mpu-6050-pinouts.jpg).

pelwell, 2020-12-15:
Never mind - the updated overlay is now in the kernel repo. You can download a test build here: https://drive.google.com/file/d/1IBUDcUCYG6JWe1nJboK6GPCoPwePDJy_/view?usp=sharing

DevynCJohnson, 2020-12-15:
Whoops, my bad. I meant to post this in raspberrypi/firmware .

Wow, you are fast. I will test the patch tonight and I will report back. Thanks!

popcornmix, 2020-12-15:
Not sure github supports transferring between Hexxeh and raspberrypi. (I'm only offered the option to transfer to Hexxeh/rpiupdate).

DevynCJohnson, 2020-12-16:
The address parameter for the MPU6050 overlay works. The output of `journalctl -xe` displays the below text, but everything still works. Thanks!

```
Dec 16 11:23:33 RPi4 kernel: inv-mpu6050-i2c 1-0069: supply vdd not found, using dummy regulator
Dec 16 11:23:33 RPi4 kernel: inv-mpu6050-i2c 1-0069: supply vddio not found, using dummy regulator
Dec 16 11:23:33 RPi4 kernel: inv-mpu6050-i2c: probe of 1-0069 failed with error -121
```

closed 2020-12-16 by DevynCJohnson