Hexxeh/rpi-firmware #254: v4l2 Failed to allocate urb buffer 0 with pwc driver (Philips webcam)

FlareP1, 2021-02-13:
v4l2 Failed to allocate urb buffer 0 with pwc driver (Philips webcam)

In the latest kernel it is not possible to capture from a Philips web cam using the pwc driver
 pwc: Philips SPC 900NC USB webcam detected.
 
 When attempting to get any output the following error/s are created
i.e.  ffmpeg -i /dev/video0 -vframes 1 output.png

 ioctl(VIDIOC_STREAMON): Cannot allocate memory
 and in DMESG
 pwc: Failed to allocate urb buffer 0
 
 On further investigation it appears to be the same as this bug, with has been diagnosed and fixed within the Linux kernel (see included patch to fix the kernel.)
 https://bugzilla.kernel.org/show_bug.cgi?id=210777
 https://bugzilla.kernel.org/show_bug.cgi?id=211195
 
 Quote from above
 "I can confirm that the PWC webcam is no longer working with the 5.10 kernel series (tried both 5.10.0 and 5.10.2) 
 but, similarly, continues to work OK with 5.9.16."
 
 I have rolled back to Rpi Kernel 5.4.83-v7l+ and can confirm that the pwc driver works again.
 
 Would it be possible to get the above solution / patch pulled into the latest Rpi Kernel?
 
 Regards

leramleram, 2021-02-13:
Same here, just ran into this after upgrading to newest software.

popcornmix, 2021-02-15:
I'm assuming fix is here: https://lkml.org/lkml/2021/1/4/772
It has a fixes tag so should appear in 5.10 kernel tree once it is merged.

FlareP1, 2021-03-10:
Update to the latest Rpi Kernel today 5.10.20-v7+and can confirm that the PWC webcam is now working again.
Thanks for the kernel updates.

closed 2021-03-10 by popcornmix