Hexxeh/rpi-firmware #255: PI 4 becomes unresponsive

sferlix, 2021-02-16:
After some hours, randomly, the PI4 becomes unresponsive. 
local mouse and keyboard not responding.
SSH and VNC no longer available.
PING still responds.

All software and firmware updated.
any help really appreciated 

thanks
frank

This is the log:

`Feb 16 07:12:09 raspSkyCam kernel: [28568.619037] ------------[ cut here ]------------
Feb 16 07:12:09 raspSkyCam kernel: [28568.619078] WARNING: CPU: 1 PID: 20140 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xec/0x128
Feb 16 07:12:09 raspSkyCam kernel: [28568.619093] Firmware transaction timeout
Feb 16 07:12:09 raspSkyCam kernel: [28568.619105] Modules linked in: fuse 8021q garp stp llc joydev brcmfmac brcmutil vc4 sha256_generic cec bcm2835_v4l2(C) v3d drm_kms_helper cfg80211 videobuf2_vmalloc gpu_sched bcm2835_codec(C) bcm2835_
isp(C) bcm2835_mmal_vchiq(C) rfkill v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 drm videobuf2_common videodev mc raspberrypi_hwmon vc_sm_cma(C) drm_panel_orientation_quirks i2c_bcm2835 snd_bcm2835(C) snd_soc_core rpi
vid_mem snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
Feb 16 07:12:09 raspSkyCam kernel: [28568.619810] CPU: 1 PID: 20140 Comm: kworker/1:3 Tainted: G        C        5.10.14-v7l+ #1401
Feb 16 07:12:09 raspSkyCam kernel: [28568.619819] Hardware name: BCM2711
Feb 16 07:12:09 raspSkyCam kernel: [28568.619837] Workqueue: events dbs_work_handler
Feb 16 07:12:09 raspSkyCam kernel: [28568.619852] Backtrace:
Feb 16 07:12:09 raspSkyCam kernel: [28568.619879] [<c0b5bafc>] (dump_backtrace) from [<c0b5be90>] (show_stack+0x20/0x24)
Feb 16 07:12:09 raspSkyCam kernel: [28568.619892]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e69fc
Feb 16 07:12:09 raspSkyCam kernel: [28568.619907] [<c0b5be70>] (show_stack) from [<c0b6021c>] (dump_stack+0xcc/0xf8)
Feb 16 07:12:09 raspSkyCam kernel: [28568.619925] [<c0b60150>] (dump_stack) from [<c0220af4>] (__warn+0xfc/0x114)
Feb 16 07:12:09 raspSkyCam kernel: [28568.619938]  r10:dec04008 r9:00000009 r8:c099ae6c r7:00000040 r6:00000009 r5:c099ae6c
Feb 16 07:12:09 raspSkyCam kernel: [28568.619948]  r4:c0e9a114 r3:c1205094
Feb 16 07:12:09 raspSkyCam kernel: [28568.619963] [<c02209f8>] (__warn) from [<c0b5c634>] (warn_slowpath_fmt+0xa4/0xd8)
Feb 16 07:12:09 raspSkyCam kernel: [28568.619974]  r7:00000040 r6:c0e9a114 r5:c1205048 r4:c0e9a134
Feb 16 07:12:09 raspSkyCam kernel: [28568.619991] [<c0b5c594>] (warn_slowpath_fmt) from [<c099ae6c>] (rpi_firmware_transaction+0xec/0x128)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620003]  r9:c1a7a540 r8:00000018 r7:00000000 r6:ffffff92 r5:c1a7a540 r4:c1205048
Feb 16 07:12:09 raspSkyCam kernel: [28568.620019] [<c099ad80>] (rpi_firmware_transaction) from [<c099af64>] (rpi_firmware_property_list+0xbc/0x170)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620030]  r7:c1205048 r6:dec04000 r5:00001000 r4:dec04024
Feb 16 07:12:09 raspSkyCam kernel: [28568.620047] [<c099aea8>] (rpi_firmware_property_list) from [<c099b088>] (rpi_firmware_property+0x70/0x118)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620059]  r10:c62b5fcc r9:00038002 r8:00000018 r7:c1a7a540 r6:c60e9d18 r5:0000000c
Feb 16 07:12:09 raspSkyCam kernel: [28568.620067]  r4:c62b5fc0
Feb 16 07:12:09 raspSkyCam kernel: [28568.620083] [<c099b018>] (rpi_firmware_property) from [<c07e3134>] (raspberrypi_clock_property+0x54/0x7c)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620095]  r10:c1ab9100 r9:29b92700 r8:2faf0800 r7:c20adac0 r6:00000000 r5:c60e9d40
Feb 16 07:12:09 raspSkyCam kernel: [28568.620105]  r4:c1205048 r3:0000000c
Feb 16 07:12:09 raspSkyCam kernel: [28568.620119] [<c07e30e0>] (raspberrypi_clock_property) from [<c07e3290>] (raspberrypi_fw_set_rate+0x4c/0xb8)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620128]  r5:c20adbc0 r4:c1205048
Feb 16 07:12:09 raspSkyCam kernel: [28568.620142] [<c07e3244>] (raspberrypi_fw_set_rate) from [<c07db658>] (clk_change_rate+0xf4/0x48c)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620153]  r7:00000000 r6:00000000 r5:00000000 r4:c20b3000
Feb 16 07:12:09 raspSkyCam kernel: [28568.620167] [<c07db564>] (clk_change_rate) from [<c07dbb6c>] (clk_core_set_rate_nolock+0x17c/0x1a4)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620178]  r10:c1ab9100 r9:00000000 r8:c20b3000 r7:2faf0800 r6:00000000 r5:c1205048
Feb 16 07:12:09 raspSkyCam kernel: [28568.620187]  r4:c20b3000
Feb 16 07:12:09 raspSkyCam kernel: [28568.620202] [<c07db9f0>] (clk_core_set_rate_nolock) from [<c07dbbd4>] (clk_set_rate+0x40/0x98)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620213]  r8:c22c1c80 r7:29b92700 r6:c1205048 r5:2faf0800 r4:c22c1c80
Feb 16 07:12:09 raspSkyCam kernel: [28568.620226] [<c07dbb94>] (clk_set_rate) from [<c09666e4>] (dev_pm_opp_set_rate+0x42c/0x588)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620236]  r5:2faf0800 r4:c22c6800
Feb 16 07:12:09 raspSkyCam kernel: [28568.620250] [<c09662b8>] (dev_pm_opp_set_rate) from [<c0971218>] (set_target+0x38/0x3c)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620262]  r10:00000000 r9:00000000 r8:00000002 r7:00000000 r6:c1404e44 r5:c1205048
Feb 16 07:12:09 raspSkyCam kernel: [28568.620270]  r4:c22c6a00
Feb 16 07:12:09 raspSkyCam kernel: [28568.620286] [<c09711e0>] (set_target) from [<c096bf1c>] (__cpufreq_driver_target+0x1bc/0x540)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620301] [<c096bd60>] (__cpufreq_driver_target) from [<c096f2ac>] (od_dbs_update+0xc4/0x170)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620312]  r10:eff26640 r9:c4747b00 r8:c49b9e80 r7:c4747900 r6:c4747b00 r5:c4747900
Feb 16 07:12:09 raspSkyCam kernel: [28568.620321]  r4:c22c6a00
Feb 16 07:12:09 raspSkyCam kernel: [28568.620337] [<c096f1e8>] (od_dbs_update) from [<c0970510>] (dbs_work_handler+0x3c/0x64)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620349]  r9:00000000 r8:c131beac r7:c22c6a00 r6:c4747904 r5:00000000 r4:c4747938
Feb 16 07:12:09 raspSkyCam kernel: [28568.620367] [<c09704d4>] (dbs_work_handler) from [<c023da60>] (process_one_work+0x254/0x584)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620379]  r9:00000000 r8:eff29900 r7:00000040 r6:eff26640 r5:c58eff80 r4:c4747938
Feb 16 07:12:09 raspSkyCam kernel: [28568.620394] [<c023d80c>] (process_one_work) from [<c023ddf0>] (worker_thread+0x60/0x5c4)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620406]  r10:eff26640 r9:c1203d00 r8:eff26658 r7:00000008 r6:eff26640 r5:c58eff94
Feb 16 07:12:09 raspSkyCam kernel: [28568.620414]  r4:c58eff80
Feb 16 07:12:09 raspSkyCam kernel: [28568.620430] [<c023dd90>] (worker_thread) from [<c0245878>] (kthread+0x170/0x174)
Feb 16 07:12:09 raspSkyCam kernel: [28568.620442]  r10:c600fe74 r9:c58eff80 r8:c023dd90 r7:c60e8000 r6:00000000 r5:c6294f40
Feb 16 07:12:09 raspSkyCam kernel: [28568.620450]  r4:c2ce07c0
Feb 16 07:12:09 raspSkyCam kernel: [28568.620464] [<c0245708>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[... 102 lines cut ...]
507395.724: Using RPI AWB
567987.341: Using RPI AWB
628208.106: Using RPI AWB
687504.957: Using RPI AWB
747694.105: Using RPI AWB
807964.708: Using RPI AWB
856893.101: TV service:host side not connected, dropping notification 0x00000001, 0x00000007, 0x00000000
865866.177: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000001
865871.022: TV service:host side not connected, dropping notification 0x00000001, 0x00000007, 0x00000000
866865.215: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000001
867875.151: Using RPI AWB
928251.367: Using RPI AWB
987515.558: Using RPI AWB
1047744.301: Using RPI AWB
1108185.468: Using RPI AWB
1168393.726: Using RPI AWB
1227674.852: Using RPI AWB
1287907.071: Using RPI AWB
1348117.310: Using RPI AWB
1408359.335: Using RPI AWB
1467589.552: Using RPI AWB
1527823.604: Using RPI AWB
1588019.589: Using RPI AWB
1648215.511: Using RPI AWB
`

pelwell, 2021-02-16:
What is typically running at the time? It looks like a camera app of some kind.

You could try adding `start_debug=1` to config.txt, then when it fails check "sudo vcbdg log msg" as above but also "sudo vcdbg log assert".

sferlix, 2021-02-16:
Yes, a camera app is running: a raspistill command is in crontab to collect an image every minute and to save it on the disk, overwriting the previous one.

About the sudo vcdbg log (posted above) there is something about camera:

`017460.432: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017761.171: camsubs: Camera not found
017761.319: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
018062.050: camsubs: Camera not found
018062.192: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
018063.820: camsubs: Camera found OK`

is this normal ?
thanks a lot
frank

sferlix, 2021-02-18:
Update:

1. new installation from scratch
2. 20 hrs with no issues
3. only thing I do is: get an image from HQ Camera every 2 minutes, by using a crontab job (raspistill <options>)
4. before launching the command, I also launch a killall raspistill, to ensure no other jobs are using the camera
5. after may hours with no issues, all of the sudden, the raspberry becomes unresponsive BUT i can still access via another PC using SSH (VNC no longer working)
6. via SSH, I issue a shutdown and I get disconnected. But the raspberry keeps showing the frozen desktop (looks like it does not shutdown)
no screen saver enabled, GPIO memory = 512, 
when it became unresponsive: temp = 52°C, CPU 9%

any help ? 
please find below the message in the log
thanks in advance
frank
`
------------[ cut here ]------------ 
Feb 18 09:42:12 raspberrypi kernel: [58182.493366] WARNING: CPU: 3 PID: 18706 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0x ec/0x128 
Feb 18 09:42:12 raspberrypi kernel: [58182.493374] Firmware transaction timeout 
Feb 18 09:42:12 raspberrypi kernel: [58182.493381] Modules linked in: rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc fuse 8021q garp st p llc joydev brcmfmac brcmutil sha256_generic cfg80211 bcm2835_codec(C) i2c_bcm2835 v3d v4l2_mem2mem rfkill bcm2835_isp(C) gpu_sched videobuf2 _dma_contig snd_bcm2835(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc4 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 cec raspberrypi_hwmon v ideobuf2_common videodev drm_kms_helper mc vc_sm_cma(C) drm drm_panel_orientation_quirks rpivid_mem snd_soc_core snd_compress snd_pcm_dmaengin e snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq nvmem_rmem uio i2c_dev ip_tables x_tables ipv6 
Feb 18 09:42:12 raspberrypi kernel: [58182.493757] CPU: 3 PID: 18706 Comm: kworker/3:3 Tainted: G        C        5.10.11-v7l+ #1399 
Feb 18 09:42:12 raspberrypi kernel: [58182.493760] Hardware name: BCM2711 
Feb 18 09:42:12 raspberrypi kernel: [58182.493785] Workqueue: events v3d_clock_down_work [v3d]
`

pelwell, 2021-02-18:
That's just telling you that the firmware stopped responding - it's just an indication that something went wrong.

As I said above:

> You could try adding start_debug=1 to config.txt, then when it fails check "sudo vcbdg log msg" as above but also "sudo vcdbg log assert".

sferlix, 2021-02-18:
sure !

I was lucky because the error now showed up quickly, after hard reboot, without having to wait for hours.

please find it below.
Hope this may help to address.
thanks 
frank

 
`
sudo vcdbg log assert
005190.840: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2014 rev 99d9a48
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
pi@raspberrypi:~/program $ sudo vcdb log msg
sudo: vcdb: command not found
pi@raspberrypi:~/program $ sudo vcdbg log msg
005194.358: arasan: arasan_emmc_open
005354.938: brfs: File read: /mfs/sd/config.txt
005355.781: brfs: File read: 1820 bytes
005449.610: brfs: File read: /mfs/sd/config.txt
006536.058: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
006538.689: *** Restart logging
006538.711: brfs: File read: 1820 bytes
006542.584: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006579.677: HDMI0: hdmi_pixel_encoding: 300000000
006579.689: HDMI1: hdmi_pixel_encoding: 300000000
006584.702: dtb_file 'bcm2711-rpi-4-b.dtb'
006590.150: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
006590.166: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xbd2d
006602.763: brfs: File read: 48429 bytes
006613.971: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
006676.967: brfs: File read: 1523 bytes
006679.169: brfs: File read: /mfs/sd/config.txt
006679.581: dtparam: i2c_arm=on
006687.981: dtparam: audio=on
006697.797: brfs: File read: 1820 bytes
006710.894: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
006727.721: Loaded overlay 'vc4-fkms-v3d'
006766.908: brfs: File read: 1446 bytes
006777.163: brfs: File read: /mfs/sd/cmdline.txt
006777.247: Read command line from file 'cmdline.txt':
006777.279: 'console=serial0,115200 console=tty1 root=PARTUUID=5239546c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
008082.439: brfs: File read: 166 bytes
009454.125: brfs: File read: /mfs/sd/kernel7l.img
009454.164: Loading 'kernel7l.img' to 0x8000 size 0x65d6a8
009454.218: Device tree loaded to 0x1ffe3c00 (size 0xc39d)
009461.697: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
014467.164: vchiq_core: vchiq_init_state: slot_zero = 0xcfd80000, is_master = 1
014470.790: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
014478.413: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
018057.734: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
018358.380: camsubs: Camera not found
pi@raspberrypi:~/program $ 
`

I also have this that sounds interesting (I hope):

`Feb 18 11:39:21 raspberrypi kernel: [    4.731063] rpivid-mem feb00000.hevc-decoder: rpivid-hevcmem initialised: Registers at 0xfeb00000 length 0x00010000
Feb 18 11:39:21 raspberrypi kernel: [    4.733784] rpivid-mem feb10000.rpivid-local-intc: rpivid-intcmem initialised: Registers at 0xfeb10000 length 0x00001000
Feb 18 11:39:21 raspberrypi kernel: [    4.735954] rpivid-mem feb20000.h264-decoder: rpivid-h264mem initialised: Registers at 0xfeb20000 length 0x00010000
Feb 18 11:39:21 raspberrypi kernel: [    4.739556] rpivid-mem feb30000.vp9-decoder: rpivid-vp9mem initialised: Registers at 0xfeb30000 length 0x00010000
Feb 18 11:39:21 raspberrypi kernel: [    4.851446] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    4.853752] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Feb 18 11:39:21 raspberrypi kernel: [    4.853779] [vc_sm_connected_init]: start
Feb 18 11:39:21 raspberrypi kernel: [    4.871244] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    4.878584] mc: Linux media interface: v0.10
Feb 18 11:39:21 raspberrypi kernel: [    4.882113] [vc_sm_connected_init]: installed successfully
Feb 18 11:39:21 raspberrypi kernel: [    4.915813] bcm2835_audio bcm2835_audio: card created with 4 channels
Feb 18 11:39:21 raspberrypi kernel: [    4.994666] bcm2835_audio bcm2835_audio: card created with 4 channels
Feb 18 11:39:21 raspberrypi kernel: [    4.998555] videodev: Linux video capture interface: v2.00
Feb 18 11:39:21 raspberrypi kernel: [    5.036313] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    5.047270] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    5.052151] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    5.064713] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    5.080318] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
Feb 18 11:39:21 raspberrypi kernel: [    5.086114] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
Feb 18 11:39:21 raspberrypi kernel: [    5.091995] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
Feb 18 11:39:21 raspberrypi kernel: [    5.096967] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[... cut ...]  snd_soc_core mc snd_bcm2835(C) vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
Feb 18 11:39:21 raspberrypi kernel: [    6.475305] CPU: 1 PID: 170 Comm: systemd-udevd Tainted: G         C        5.10.11-v7l+ #1399
Feb 18 11:39:21 raspberrypi kernel: [    6.475320] Hardware name: BCM2711
Feb 18 11:39:21 raspberrypi kernel: [    6.475335] Backtrace: 
`

pelwell, 2021-02-18:
Is that with the camera attached? The firmware says it can't find it, which isn't a good sign.

sferlix, 2021-02-18:
yes, the camera properly it's attached. No one touched the camera and cable.

I have tried with 3 different cameras: official HQ, official NoIR and unofficial HQ. Same issue (not systematic, but happens randomly)..
I also changed cables (actually, each camera has its own cable)

Please note that when the Pi becomes unresponsive, I have to switch off.
Once I restart the Pi, it works fine without moving the camera (until next failure).

Could it be an issue with the board ? Maybe I could return the board, as I bought it on Amazon 1 week ago.
thanks
frank

pelwell, 2021-02-18:
I rarely use the cameras, but it should be detected correct on each boot.

How is your system powered? Do you have any other peripherals attached?

In one of the earlier posts it says `GPIO memory = 512,` - I presume that's really talking about `gpu_mem`. Does it need to be such a large carve-out for the VPU?

sferlix, 2021-02-18:
system is powered by power cable (the one provided by the case manufacturer)

sorry, it was gpu_mem = 512
formerly it was 256 and also 128. I decided to increase it after having experienced issues.
I go back to 256

thanks

sferlix, 2021-02-18:
Ciao,

while using the camera via command line (no crontab) to detect optimal parameters, I got into this:

ERROR: <class 'picamera.exc.PiCameraRuntimeError'>
Traceback (most recent call last):
  File "skycam.py", line 137, in <module>
    camera.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
###     ****"No data recevied from sensor. Check all connections, "**
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board**

after this, I start getting:

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
ERROR: <class 'picamera.exc.PiCameraMMALError'>

So it seems you are rigth. But cables are well connected. Issue happens with all cameras I have tried.
Should I return the Pi ?

thanks
frank

sferlix, 2021-02-19:
Again after 20ish hours.

Here is the log (many of these messages)

`543438.771: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev 99d9a48                                                                                                                    vcdbg_ctx_get_dump_stack: dump_stack failed                                                                                                    ----------------                                                                                                                               3663110.578: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev 99d9a48                                                                                                          vcdbg_ctx_get_dump_stack: dump_stack failed `

And this 

`37788.845: Using RPI AWB                                                                                                                     3657829.036: imx477_open: read id 1143 succes 0!                                                                                               3657842.042: Using RPI AWB                                                                                                                     3665828.315: mmal: fail_destroy: Timed out waiting to destroy ril.camera                                                                       1907489.263: mmalsrv: mmal_server_create_control_service: failed to create control service - error 1                                           2313809.362: TV service:host side not connected, dropping notification 0x00000001, 0x00000007, 0x00000000                                      2315783.985: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000001                                      2322823.500: TV service:host side not connected, dropping notification 0x00000001, 0x00000007, 0x00000000                                      2323893.713: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000001                `

Looks the same...

pelwell, 2021-02-19:
Is that the same Pi 4 or a different one?

sferlix, 2021-02-19:
same Pi.
Sorry, I just realized that some log messages are not readable..

PMLC-Mike, 2021-02-23:
I think I am seeing similar (not exactly same) issue and can reproduce the outcome via the following steps. Hopefully this helpful and not a different matter.

Apologies in advance for any reporting faux pas. This is my first GitHub post/comment.

I am running Raspberry Pi 3 Model B Plus Rev 1.3 with latest Raspian upgrades/updates as of today along with UFW and fail2ban installed. Using Pi Camera V2.

PART 1 ---------------------------------------------------------------------------------------------------

I was trying to get the following (test.py code below) to work but kept getting the memory error below... (note that the code works without memory error if camera default resolution 1920x1080 is used instead of 3264x2464)

After boot, via terminal...

$ cd scanner
$ source venv/bin/activate
(venv) $ pip list

Package               Version 
--------------------- --------
cycler                0.10.0  
decorator             4.4.2   
imageio               2.9.0   
imutils               0.5.4   
kiwisolver            1.3.1   
matplotlib            3.3.4   
networkx              2.5     
numpy                 1.20.1  
opencv-contrib-python 4.1.0.25
picamera              1.13    
Pillow                8.1.0   
pip                   18.1    
pkg-resources         0.0.0   
pyparsing             2.4.7   
python-dateutil       2.8.1   
PyWavelets            1.1.1   
scikit-image          0.18.1  
scipy                 1.6.0   
setuptools            40.8.0  
six                   1.15.0  
tifffile              2021.2.1

(venv) $ cat test.py

from picamera.array import PiRGBArray
from picamera import PiCamera
import time

print("Code run started")

camera = PiCamera()
camera.resolution = (3264,2464)

camera.start_preview()
time.sleep(2.0)
camera.stop_preview()

raw_capture = PiRGBArray(camera)
time.sleep(2.0)
camera.capture(raw_capture, format="bgr")
image = raw_capture.array

print("Code run completed.")

(venv) $ python test.py
Code run started
mmal: mmal_vc_port_enable: failed to enable port vc.ril.camera:out:2(BGR3): ENOMEM
mmal: mmal_port_enable: failed to enable port vc.ril.camera:out:2(BGR3)(0x19e0640) (ENOMEM)
Traceback (most recent call last):
  File "/home/rpi/scanner/venv/lib/python3.7/site-packages/picamera/camera.py", line 1418, in capture
    encoder.start(output)
  File "/home/rpi/scanner/venv/lib/python3.7/site-packages/picamera/encoders.py", line 1166, in start
[... 83 lines cut ...]
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629218] [<8013b538>] (process_one_work) from [<8013bb1c>] (worker_thread+0x60/0x5c4)
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629225]  r10:af8e8640 r9:80e03d00 r8:af8e8658 r7:00000008 r6:af8e8640 r5:81650014
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629229]  r4:81650000
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629238] [<8013babc>] (worker_thread) from [<801435a4>] (kthread+0x170/0x174)
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629245]  r10:81565e74 r9:81650000 r8:8013babc r7:8168c000 r6:00000000 r5:8164d040
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629249]  r4:8164d100
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629256] [<80143434>] (kthread) from [<801000ec>] (ret_from_fork+0x14/0x28)
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629262] Exception stack(0x8168dfb0 to 0x8168dff8)
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629267] dfa0:                                     00000000 00000000 00000000 00000000
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629275] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629280] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629287]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143434
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629291]  r4:8164d040
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629296] ---[ end trace 38f990ef8c8483eb ]---
Feb 22 17:21:16 raspberrypi-scanner kernel: [  172.629326] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 4294967186 (-34)
Feb 22 17:21:17 raspberrypi-scanner kernel: [  173.668493] hwmon hwmon1: Failed to get throttled (-110)
Feb 22 17:21:18 raspberrypi-scanner kernel: [  174.708449] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -110

PART 3 -----------------------------------------------------------------------------------------------

sudo reboot now does not reboot machine; have to unplug and plug back in to unfreeze screen, etc.

Resetting GPU memory to 128MB returns back to PART 1 above.

Of side note: I was not having this problem on Pi Zero W a week ago. Only encountered when I moved over to Pi 3B+

PMLC-Mike, 2021-02-23:
PS to comment above: camera confirmed working via raspistill command.

PMLC-Mike, 2021-02-23:
PPS. I forgot to mention that the above occurs when I am remotely connecting to the Pi using VNC (with Direct Capture Enabled).

zt4, 2021-02-23:
I've got the same issue... one Pi 4 with a camera, two Pi Zero W's with a camera. I run raspistill in a loop and take an image around 4/minute. All of them hang up after anywhere from a few hours to a day or two. On the Pi 4, I stopped the camera script that I was collecting images, and it does not crash. If I catch it soon enough, I can SSH in and reboot, although it typically hangs on reboot anyway. Otherwise it will eventually stop responding to ping or anything.

This all started somewhere around the 5.10 update.

I have a couple of other Pi's with cameras that do not hang up, and I don't know what would be different with them. They were all updated about the same day to the 5.10 update, but a 3B and another Zero W do not have any issues with the camera. I've updated the troubled Pi's with rpi-update to try to fix, but the updates haven't made any difference.

sferlix, 2021-02-23:
what camera are you using ?
thanks
francesco

sferlix, 2021-02-23:
I am running another test.
on a new Pi4, new SD, i have installed a Raspberrian Lite (no desktop).
the very same script is now running since 1,5 days with no issues,
collecting images every 2 minutes.
I will keep testing for more days.

important to know: I did not run any upgrade, while, in the former Pi4, I
upgraded the firmware.

Yes, it's a different hardware and a different OS (no desktop). but I have
the impression that the errors started right after the firmware upgrade...

who knows.. i hope to pass the tests...
francesco

zt4, 2021-02-23:
I'm using:

Official camera v2 on the two Pi Zero's (locks up).
Official camera v2 on two Pi 3B's (works fine).
Official camera v2 on a Pi 3B+ (works fine).
Waveshare fisheye camera (with I think are just v1 with a different lens) on a Pi4 (locks up).
Waveshare fisheye camera on a Pi Zero (works fine).

sferlix, 2021-02-23:
I have the same ìssue with V2 Camera and HQ Camera as well on the Pi4
(upgraded firmware and Raspbian desktop)

Now testing with HQ Camera, new Pi4 (no upgraded firmare and Raspbian Lite,
same cables). As said, no issues.

thanks
francesco

sferlix, 2021-04-28:
see last message

closed 2021-04-28 by sferlix