Hexxeh/rpi-firmware #256: Plymouth script module based themes broken

devshanky, 2021-02-22:
Facing issues with Plymouth script based themes on RaspiOS 64-bit arm64 updated to latest firmware. It may randomly show up during boot and sometimes it flashes and disappears. Mostly observed when the screen resolution changes from 720p to 1080p or vice-versa.
 
2-step throbber based themes works fine.

-R

devshanky, 2021-02-22:
Always works with hdmi_safe=1.

It was working reliably on 5.4.79-v8+ without hdmi_safe=1.

-R

popcornmix, 2021-02-22:
Does switching
```
dtoverlay=vc4-fkms-v3d
```
to
```
dtoverlay=vc4-kms-v3d
```
in config.txt help?

devshanky, 2021-02-22:
No. Tried both.

-R

devshanky, 2021-02-22:
I have a feeling it may have to do with **initramfs**. 

I am able to consistently see the plymouth screens if I do **update-initramfs -c -k $(uname -r)** after a change in screen resolution. However I lose it again on subsequent reboot. I am unable to understand this behavior. It is not desirable to **update-initramfs** after change in screen resolution.

I do **xrandr -s** for immediate effect and also update **config.txt** with **hdmi_mode** and **hdmi_group** with appropriate values.

-R

devshanky, 2021-02-22:
@popcornmix **WELL WELL WELL....** 

I commented out **dtoverlay=vc4-fkms-v3d** and now the things are back to normal.

I compared output of **dmesg** from both the scenarios when things work good and when it doesn't with dtoverlay=vc4-fkms-v3d enabled. 

Looks like a timing issue with loading overlay drm driver. Please look a the image attached with this message.

[image]

Unfortunately, I will not be able to use kms or fkms driver with this update.

-R