Hexxeh/rpi-firmware #257: Cirrus Logic Audio Card: Recording stutters

choff, 2021-02-22:
I have a Cirrus Logic Audio Card connected to a Raspberry Pi 4 and am facing a really strange issue since my last dist-upgrade: sometimes, recording audio from the sound card stutters and there are "gaps" in the recording. Also "arecord" reports overflow errors:
```
 $ arecord 1kHz-broken.wav
Recording WAVE '1kHz-broken.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
overrun!!! (at least 0.086 ms long)
overrun!!! (at least 0.111 ms long)
overrun!!! (at least 0.102 ms long)
overrun!!! (at least 0.108 ms long)
overrun!!! (at least 0.101 ms long)
overrun!!! (at least 0.115 ms long)
```
I have connected the 1 kHz tone generator of the Cirrus Logic Audio Card to the audio input path. The resulting (broken) recording is [provided here](https://github.com/Hexxeh/rpi-firmware/files/6025035/recordings.zip). You cannot hear the "gaps" during the recording as silence as "arecord" just seems to "jump" over the gaps. But what you can hear is that the 1 kHz audio is distorted. Th "gaps" in the recording (which unfortunately you cannot hear) are so bad that I cannot be understood by anyone when I am in a phone call. A good recording that doesn't show these issues is also in the same ZIP archive as the broken recording.

I have nailed this issue down to [this commit](https://github.com/Hexxeh/rpi-firmware/commit/43022f5adbbf5d6e05ea0e022a3090c2c9feff7c): "firmware: dmalib: Allow sdcard to borrow channel 6". Every firmware version before this commit works well. Starting from this commit, every firmware version (even the most recent one) shows this issue. I did a lot of testing of older firmware revisions with "sudo rpi-update [commit ID]", so this is why I am so sure. I also tested the same thing with another Cirrus Logic Audio Card and this one behaves in the same way. So this cannot be a software issue.

My test vehicle is a Raspberry Pi 4 without SD card (it has an external SSD connected via USB). I don't know if the Raspberry Pi 3 and other models are also affected, but I do have a Raspberry Pi 3, so I could give it a try.

If the issue happens and my colleagues complain about the stuttering, it often disappears after a reboot. Likewise, a reboot can also cause the stuttering to appear if it wasn't there before the reboot. So, the stuttering is a sporadic issue. It may happen or it may not, but if it happens then it keeps on happening in every recording until you reboot the system. And if it works, all recordings are fine until you reboot the system.

Do you have any clue how this change "firmware: dmalib: Allow sdcard to borrow channel 6" could cause this kind of failure? Do you have any ideas what I could still try out?

Thank you.

popcornmix, 2021-02-23:
Can you move this report to https://github.com/raspberrypi/linux/issues where it belongs?

closed 2021-02-23 by popcornmix

choff, 2021-02-23:
@HiassofT, could you please take a look at this issue and let us know if you have any idea what might be happening here?

popcornmix, 2021-02-23:
No firmware bug reports go to https://github.com/raspberrypi/firmware/issues
As long as it's in the raspberrypi repo we can move issues between linux and firmware - we can't for issues filed here.

HiassofT, 2021-02-23:
I did a quick test and couldn't reproduce the issue - playback and recording worked fine (using 5.10.16 kernel and Feb 16 firmware) on a netbooted RPi4

choff, 2021-02-23:
Thanks a lot @popcornmix and @HiassofT for your help! I have moved the issue to [https://github.com/raspberrypi/firmware/issues/1541](https://github.com/raspberrypi/firmware/issues/1541) as you suggested.

@HiassofT: hum, strange that you cannot reproduce it... Did you try booting multiple times? Recording only fails after every 2nd boot or so. So, I always boot the system 5 times and try to reproduce it before I am sure that a particular firmware version is good and doesn't show this issue. Do you have any idea how a firmware change entitled "firmware: dmalib: Allow sdcard to borrow channel 6"] could break the recording? Does the kernel rely on DMA functionality in the firmware for I2S playback and recording?