Hexxeh/rpi-firmware #260: snd_soc_hifiberry_dacplus no longer loads

lbt, 2021-03-11:
After an update installed the latest kernel (5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021) my Hifiberry Dac+ did not work.

I tried a plain "rpi-update" which gave me 5.10.20-v7l+ #1404 SMP Thu Mar 4 19:44:07 GMT 2021 with the same issue.
snd-rpi-hifiberry-dacplus was not mentioned in the bootup/dmesg

Downgrading to 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
using:
  rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2
Gave me a working dac+ with entries in aplay -L and dmesg had:
snd-rpi-hifiberry-dacplus soc:sound: pcm512x-hifi <-> fe203000.i2s mapping ok

Finally I tried (on the same standard 'buster' install):
  apt-get install --reinstall raspberrypi-kernel
This gave 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
I wanted to check I'd not done anything silly and indeed - no dac+ in dmesg or aplay -L

pelwell, 2021-03-12:
It's working for me:
```
pi@raspberrypi:~ $ dmesg | grep hifi
[    6.421693] snd-rpi-hifiberry-dacplus soc:sound: ASoC: no DMI vendor name!
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0 [HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
It appears to be playing OK, although there's nothing hooked up to the phono outputs to prove it.

With the dacplus removed but with the overlay still applied the kernel reports:
```
[    6.462015] pcm512x 1-004d: Failed to reset device: -121
```

Please post the output of `aplay -l`, `sudo vcdbg log msg` and `dmesg -l err`.

lbt, 2021-03-12:
You mention "but with the overlay still applied" ... I do not use an overlay in config.txt at the moment (it seems to be auto-detected). What dtoverlay line do you have?
Maybe the issue will become autodetect not working :)

pelwell, 2021-03-12:
I have a rev 2.2 which lacks a HAT EEPROM - perhaps your board is an earlier revision.

The `sudo vcdbg log msg` command will show the firmwares attempt to read (or not) the HAT EEPROM. Also, the firmware creates /proc/device-tree/hat if one exists and populates it with details of the board.

In the absence of a (working) EEPROM the required dtoverlay line is `dtoverlay=hifiberry-dacplus`. If the problem is that the built-in overlay is present but not working you can disable it by adding `dtoverlay=` as the very first "dt*" line in config.txt.

lbt, 2021-03-12:
Edited the issue and noted that this is a Pi 4
My board is rev 2.6 (approx mid-2019)

OK, so with dtoverlay=hifiberry-dacplus on 5.10.20-v7l+ #1404 SMP Thu Mar 4 19:44:07 GMT 2021 it works.

Some key words helped me find this which explains what's happening :
 https://www.hifiberry.com/blog/configuration-changes-in-linux-5-4/

Closing - thanks very much for your help :)

closed 2021-03-12 by lbt