Hexxeh/rpi-firmware #262: Kernel headers for downgraded kernel

Artnal, 2021-04-01:
Hi there, 

Due to a [specific issue](https://github.com/raspberrypi/linux/issues/3981) I've had to downgrade my kernel version for raspberry pi to 4.19.118-v7l. In doing so, I broke my wireguard install, or at least the part of it that relied on the kernel headers. At present, it seems like the kernel is properly downgraded, but the kernel headers are still on their latest version. 

I tried 
`sudo apt-get install raspberrypi-kernel-headers`, which results in 
> Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20210303-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded

`sudo apt-get install raspberrypi-kernel-headers-4.19.118-v7l`, however, results in 
>Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers-4.19.118-v7l
E: Couldn't find any package by glob 'raspberrypi-kernel-headers-4.19.118-v7l'
E: Couldn't find any package by regex 'raspberrypi-kernel-headers-4.19.118-v7l'

Would anyone have a suggestion to downgrade the kernel headers to match the kernel version currently installed? 
Thanks!

popcornmix, 2021-04-01:
https://github.com/RPi-Distro/rpi-source may help

Artnal, 2021-04-03:
Thanks for the suggestion. I gave it a shot and ran `rpi-source` but at the end of the process, the issue remains. I am not sure I am knowledgeable enough in linux to explore further with this tool atm.