Hexxeh/rpi-firmware #264: Firefox youtube not working after flashing to the latest version firmware on Raspberry Pi 4B

mrusli70, 2021-04-19:
After flashing to the latest version for the Raspberry Pi 4, firefox not working with youtube. 

Built-in Audio Digital Stereo	/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0		Enabled	All	default: S16LE, support: S16LE S16BE F32LE F32BE	2	default: 44100, support: 1 - 384000	0 - 0
Built-in Audio Analog Stereo	/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card1		Enabled	None	default: S16LE, support: S16LE S16BE F32LE F32BE	2	default: 44100, support: 1 - 384000	0 - 0
Input Devices
Name 	Group 	Vendor 	State 	Preferred 	Format 	Channels 	Rate 	Latency
Monitor of Built-in Audio Digital Stereo	/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0		Enabled	All	default: S16LE, support: S16LE S16BE F32LE F32BE	2	default: 44100, support: 1 - 384000	0 - 0
Monitor of Built-in Audio Analog Stereo	/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card1		Enabled	None	default: S16LE, support: S16LE S16BE F32LE F32BE	2	default: 44100, support: 1 - 384000	0 - 0

popcornmix, 2021-04-19:
Be more specific. What was "latest version for the Raspberry Pi 4"? e.g. a download link.
And are you saying you installed apt installed firefox (with no other changes or updates) and audio doesn't work?
HDMI or analogue?
Does other audio work? e.g. VLC or chromium?

mrusli70, 2021-04-19:
I flash the firmware to the latest version to 5.10.27 using rpi-update and i do not how the firefox cannot seems to work. I formated my SD Card and reinstall raspbianOS. Still the same thing on firefox no audio. I am on HDMI. Chromium and VLC works fine. Only on Firefox.

popcornmix, 2021-04-19:
> I flash the firmware to the latest version to 5.10.27 using rpi-update and i do not how the firefox cannot seems to work. I formated my SD Card and reinstall raspbianOS.

Did it work after reinstalling raspbianOS but without running rpi-update?

mrusli70, 2021-04-19:
formatted the SD CARD, and reinstall raspbianOS fresh still, forefox did not work. Only on Chromium. Even did not do rpi-update.

popcornmix, 2021-04-19:
This won't be a firmware issue - it sounds like either a firefox or pulseaudio issue.
I'll close this but you may want to continue report here: https://github.com/RPi-Distro/repo/issues/218

closed 2021-04-19 by popcornmix