Hexxeh/rpi-firmware #265: Kernel headers (source) for 5.10.46

bluuu81, 2021-06-29:
If we want to add any module to kernel we need latest headers and source.
For example for dkms and cm4io-fan ... I can't build module
Any advices are welcome

popcornmix, 2021-06-29:
See https://github.com/RPi-Distro/rpi-source

bluuu81, 2021-06-29:
Yeah, thanks
Now module is building ... but can't work because of modprobe: ERROR: could not insert 'emc2301': Exec format error (with and without --force)
Don't know is a module or headers issue :/

pelwell, 2021-06-29:
What is your module `make` command line? And the output of `uname -a`?

bluuu81, 2021-06-29:
Makefile (so make too is from cm4io-fan: https://github.com/neg2led/cm4io-fan):
https://github.com/neg2led/cm4io-fan/blob/master/Makefile

uname -a:
Linux greennas 5.10.46-v8+ #1430 SMP PREEMPT Mon Jun 28 17:17:17 BST 2021 aarch64 GNU/Linux

pelwell, 2021-06-30:
That Makefile doesn't make it obvious what the actual kernel make line would be. For example, does it include `ARCH=aarch64`?
I suspect your problem might be that you are trying to build a 64-bit kernel with a native 32-bit gcc. What does `file vmlinux` return after the build?