Hexxeh/rpi-firmware #266: No signal TV kernel 5.10.46

vlad34, 2021-07-03:
Hi,

I updated kernel Raspberrypi OS to 4.10.46 on my raspberry pi 4 (4Go), the reboot cause no signal (not screen) for the TV in console boot mode. (HDMI connection)

I downgrade kernel to 4.10.44 , it's works the signal on the TV.

The problem is with upgrading to kernel 4.10.46.

Good day.

popcornmix, 2021-07-04:
Are you using the kms driver? Any other non-default settings?
When working what resolution/framerate are you using on display?
Can you ssh in when showing no signal and get`dmesg` output?
Can you identify the first commit with the issue (using `rpi-update <hash>` with hash from https://github.com/Hexxeh/rpi-update/commits

pelwell, 2021-07-04:
Please include the content of `/boot/cmdline.txt`.

vlad34, 2021-07-04:
I reinstalled the 5.10.46 kernel with rpi-update and I have the problem again

@popcornmix,

I don't use the KMS driver, I run on the legacy driver, I think...

1920x1080 , 60f

Get dmesg output [https://pastebin.com/ybW2uHer](url)

Hash first commit issue : 23fd671886ba071e04e9787835a05dfc84137679

My config /boot/config.txt [https://pastebin.com/rktAh5NE](url)

@pelwell

cmdline.txt:

`dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=330c27f0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

Thanks,

alanbork, 2021-07-04:
you are running fkms.  this is probably a dupe of https://github.com/raspberrypi/linux/issues/4428.

popcornmix, 2021-07-05:
Can you try with latest rpi-update kernel?

vlad34, 2021-07-05:
@popcornmix

It's works! 

Thanks. 😁

closed 2021-07-05 by vlad34