Hexxeh/rpi-firmware #267: HDMI-CEC not working

HITMIK, 2021-07-10:
Hi,

i have samsung and rpi4 with newest rpi-firmware. On kodi not work cec. 

I try unpluge samsung and rpi4 power on 10 minutes.

Any solution?

popcornmix, 2021-07-10:
I assume you ran `rpi-update`? Was cec working before you ran it?

HITMIK, 2021-07-12:
Yes, cec run. But i dont know since when cec dont work (how version crash it). Maybe problem with raspberry firmware? Maybe your fw is OK.

popcornmix, 2021-07-12:
Sorry, you've not provided enough information for any useful support.
Please explain exactly what sdcard image you are running and if CEC previous ran, what happened to stop it working (e.g. did you upgrade any software).

e-losin, 2021-07-30:
Hello!
I can confirm that - hdmi-cec does not work with "fresh" firmware.

For test, i done some checks on fw versions:

I. hdmi-cec not working on:
Linux raspi 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux
Linux raspi 5.10.50-v7l+ #1437 SMP Thu Jul 15 11:48:49 BST 2021 armv7l GNU/Linux
Linux raspi 5.10.49-v7l+ #1436 SMP Wed Jul 14 14:18:38 BST 2021 armv7l GNU/Linux

II. hdmi-cec working correctly on:
Linux raspi 5.10.46-v7l+ #1430 SMP Mon Jun 28 17:14:36 BST 2021 armv7l GNU/Linux

All test was made on one TV/cables etc. HDMI-CEC functionality was tested on kodi 18.7
My raspi:
```
root@raspi:/home/pi# lsb_release -a                    
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
root@raspi:/home/pi# 

root@raspi:/home/pi# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@raspi:/home/pi# 

```

popcornmix, 2021-07-30:
Thanks that helps. I assume you are using the fkms driver?
Can you confirm if https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2 is working or not?

e-losin, 2021-07-31:
> Thanks that helps. I assume you are using the fkms driver?
> Can you confirm if [9dbcd6d](https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2) is working or not?

I assume you are using the fkms driver?
```
root@raspi:/home/pi# cat /boot/config.txt | grep fkms
dtoverlay=vc4-fkms-v3d
```

[9dbcd6d](https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2)
No, on this commit hdmi-cec not working.

HITMIK, 2021-08-01:
Yes men!

On "Linux raspi 5.10.46-v7l+ #1430 SMP Mon Jun 28 17:14:36 BST 2021 armv7l GNU/Linux" HDMI-CEC WORKINS!

popcornmix, 2021-08-02:
Can you test with lastest rpi-update firmware which has a potential fix.

e-losin, 2021-08-02:
With commit ce56fb43ea36888b94026edd1f08aea78cf32262 it works! 
Thank you!