Hexxeh/rpi-firmware #268: Black screen , commit "b986fda889737f6ce0a4acda89d447f78d0a1a73"

vlad34, 2021-07-30:
Hi,

I updated kernel Raspberrypi OS to b986fda889737f6ce0a4acda89d447f78d0a1a73 on my raspberry pi 4 (4Go), the boot cause black screen for the TV in console boot mode. (HDMI connection)

I downgrade commit to 1fad1d2beceaaf96e84e3f5caa79f52b87332e89 , it's works.

The problem is with upgrading to commit b986fda889737f6ce0a4acda89d447f78d0a1a73 .

Use the KMS driver.

Resolution: 1920x1080@60

Get dmesg output: https://pastebin.com/c3NVKiaB

Hash first commit issue : b986fda889737f6ce0a4acda89d447f78d0a1a73

My config /boot/config.txt: https://pastebin.com/shBvGPQB

$ lsb_release -a

`
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
`

cmdline.txt:

`
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=330c27f0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
`

Good day.

timg236, 2021-07-30:
Please can you post the contents of config.txt and the SDRAM size. If a large gpu_mem configuration is specified then this will conflict with the extra CMA memory. If so, the fix is to reduce gpu_mem because that GPU won't be using that memory

timg236, 2021-07-30:
Missed the config.txt link. There's no benefit in specifying 512MB of GPU memory because only the frame buffer and codec buffers are allocated in gpu mem. 256 is probably a reasonable maximum

popcornmix, 2021-07-30:
> 256 is probably a reasonable maximum

on 5.10 kernel there's rarely a need for more than default (76M if unspecified) gpu_mem.
3d and hevc buffers no longer use gpu mem.

vlad34, 2021-07-30:
I had it configured to 512MB GPU because I am using kodi and saw that I have 4GB of usable RAM.

I just cut it down to 256MB and it's working.

Thanks.

closed 2021-07-30 by vlad34

popcornmix, 2021-07-30:
I'd suggest removing the line completely - it's just reducing the amount of usable memory.
The default (76M) is enough for Blu-Ray quality H.264.