Hexxeh/rpi-firmware #271: Waveshare 7" HDMI LCD (C): touch doesn't work properly anymore

meandthemachine, 2021-09-29:
Hello,

I'm not sure if this is a kernel issue, but maybe someone could classify this and/or give me a hint for a solution or the cause.

I've got the following setup:
- Raspberry Pi 3 model B+
- Waveshare 7" HDMI LCD (C) Rev 2.1 (https://www.waveshare.com/wiki/7inch_HDMI_LCD_%28C%29), it identifies as USB device "BYZHYYZHY By ZH851"
- Latest Raspberry Pi OS/Raspbian Buster (as of 2021-05-07)

In the given setup, the touch recognition doesn't work properly. For example, on the desktop, the mouse cursor jumps to the tapped spots on the screen, but:
- taps are not detected as clicks, several taps/attempts are needed till click is recognized and executed
- it randomly switches to drag-and-drop mode when tapped, one needs to press ESC to leave this mode
- it randomly switches to selection mode (squares drawn by mouse cursor) when tapped, ESC leaves

So, starting from latest Raspberry Pi OS version mentioned above, I used `rpi-update` to go back in kernel versions in intervals until I found that touchscreen recognition worked properly again with kernel version [41363960f4fe1029664ac393dc6b2076df166548](https://github.com/Hexxeh/rpi-firmware/commit/41363960f4fe1029664ac393dc6b2076df166548). If I use one commit later ([63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8](https://github.com/Hexxeh/rpi-firmware/commit/63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8)) and any version afterwards, the described effects can be observed, again.

I also observed different behaviour of latest Raspberry Pi OS release with shipped kernel, when I use different hardware:
- touch works just fine if I replace Raspberry Pi 3B+ by Rapsberry Pi 4B and keep everything else in place
- touch works fine if I replace the LCD by another examplar of that exact model (and keep everything else in place)

Regarding the LCDs: I own three exemplars of that exact model Waveshare 7" HDMI LCD (C) Rev 2.1. 
When executing `dmesg | grep input` the one in question identifies as "BYZHYYZHY By ZH851", whereas the two other ones identify as "WaveShare WaveShare". So there seems to be a different firmware version installed, although they all are "Rev 2.1"...
Model Rev 2.1 is said to support standard HID: https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)#About_the_version

I already contacted Waveshare support and even sent them a video, but their reply was not very helpful and I didn't learn anything new:
> Since the 7inch HDMI LCD (C) Rev 2.1 is an old version of this model, maybe it cannot compatible well with the new kernel.
If you need to use the new version of the Raspberry Pi OS, you can try to use our new version of this LCD

This would render the LCD into electronic waste, which is not an option for me.

I'm fine with keeping my current solution/workaround (which is using kernel version [41363960f4fe1029664ac393dc6b2076df166548](https://github.com/Hexxeh/rpi-firmware/commit/41363960f4fe1029664ac393dc6b2076df166548)), but as said in my statement at the beginning, maybe somebody can give me a hint for a more "updateable" solution or for the cause, by comparing the differences between the given kernel versions.

Thanks a lot and cheers!

popcornmix, 2021-09-29:
Can you create this issue here:
https://github.com/raspberrypi/linux/issues
where the right people are more likely to see it.

closed 2021-09-29 by popcornmix

meandthemachine, 2021-09-29:
@popcornmix: Just to be sure. Didn't you mean the official firmware repo
https://github.com/raspberrypi/firmware/issues
?

popcornmix, 2021-09-29:
No. Firmware has no knowledge of USB devices. They are handled by the linux kernel.
And the rpi-update hash you identified contained purely a single kernel commit and no firmware changes.

meandthemachine, 2021-09-29:
OK, thank you for your explanation. I know far too less about all of this...