Hexxeh/rpi-firmware #272: Playback of an RTSP stream no longer works via VLC player

Blackbox-git, 2021-10-17:
## Issue

Up to the rpi-firmware (stable-branch) mentioned below, I was able to display the stream of my network camera using the following command. If I update the firmware to a newer one, the VLC player starts, but no picture appears.

## The command

`DISPLAY=:0 vlc rtsp://Username:Password@IP-Adress/11--quiet --rtsp-tcp --noaudio --fullscreen --no-qt-privacy-ask --no-video-title --network-caching=500 --sout-mux-caching=500
`

## Addtion
The problem is reproducible on a second Raspberry Pi.

Blackbox-git, 2021-10-17:
[image]

Blackbox-git, 2021-10-17:
Okay, ich will do this tomorrow. If i am reday i will give feedback.

Blackbox-git, 2021-10-18:
[image]

popcornmix, 2021-10-18:
Thanks, looks like a duplicate of https://github.com/RPi-Distro/vlc/issues/48

Does your rtsp stream change resolution?
We may be interested in accessing the stream if we don't find a simpler way of reproducing issue.

Blackbox-git, 2021-10-18:
The connected screen has 10" and a resolution of 1024 x 600. This is stored accordingly in config.txt. Until this update, everything looked perfect.

The stream itself has a resolution of 1920x1080. I think, but the behaviour is also with a Full-HD monitor.

I have opened a separate DNS address for you to test. The IP is updated from a Raspberry. 

Here is the command line command for the stream:

DISPLAY=:0 vlc rtsp://guest:testcam@testcam.dnsuser.de:554/11 --quiet --rtsp-tcp --noaudio --fullscreen --no-qt-privacy-ask --no-video-title --network-caching=500 --sout-mux-caching=500

6by9, 2021-10-18:
Thank you.

As I've just noted on https://github.com/RPi-Distro/vlc/issues/48 I've reproduced the issue against the Youtube link they provided there. Testing the fix against your stream also appears to work, so it looks to be good.
The patch is now pushed for internal review, and then release.

pelwell, 2021-10-19:
1. It will get there eventually, when we decide that enough has changed to make it worthwhile or that sufficient time has elapsed since the last update.
2. You don't need to take any special action - both update mechanism are unaware of the other and will happily overwrite files installed by the other,