Hexxeh/rpi-firmware #274: /lib/modules/5.10.81+/build The file or directory does not exist.

AzagraMac, 2021-11-25:
After doing a `rpi-update`, I updated my RaspberryPi, to the 5.10.81+ kernel, but I find that the build directory does not exist in this kernel, but it does exist in the 5.10.63+ kernel.

To fix this, I had to do a downgrade with: `sudo apt install --reinstall raspberrypi-bootloader raspberrypi-kernel`

popcornmix, 2021-11-25:
Did you previously build your own kernel from 5.10.63 source?
What does
```
ls -l /lib/modules/5.10.63+/build
```
report. Why do you want a build directory? What are you trying to do with it?

AzagraMac, 2021-11-25:
I have some packages to compile, and they need the kernel source... I have to go back to the old kernel to do it, the new one does not contain those sources. 

```
$ ls -l /lib/modules/5.10.63+/build
lrwxrwxrwx 1 root root 31 oct 29 12:31 /lib/modules/5.10.63+/build -> /usr/src/linux-headers-5.10.63+
```

`Install v4l2loopback from source] ****************************************************
fatal: [XXX.XXX.XXX.XXX]: FAILED! => {"changed": false, "cmd": "cd /tmp/v4l2loopback-0.12.5 && make && sudo make install && depmod -a", "delta": "0:00:00.104160", "end": "2021-11-25 16:39:16.782732", "msg": "non-zero return code", "rc": 2, "start": "2021-11-25 16:39:16.678572", "stderr": "make[1]: *** /lib/modules/5.10.81+/build: No such file or directory.  Alto.\nmake: *** [Makefile:43: v4l2loopback.ko] Error 2", "stderr_lines": ["make[1]: *** /lib/modules/5.10.81+/build: No existe el fichero o el directorio.  Alto.", "make: *** [Makefile:43: v4l2loopback.ko] Error 2"], "stdout": "Building v4l2-loopback driver...\nmake -C /lib/modules/`uname -r`/build M=/tmp/v4l2loopback-0.12.5 modules", "stdout_lines": ["Building v4l2-loopback driver...", "make -C /lib/modules/`uname -r`/build M=/tmp/v4l2loopback-0.12.5 modules"]}`

popcornmix, 2021-11-25:
You might want to check out [rpi-source](https://github.com/RPi-Distro/rpi-source)

The headers from apt will match versions of kernel supplied through apt.
If you are using rpi-update test kernels, you need to get corresponding headers from elsewhere.

AzagraMac, 2021-11-25:
The only thing that is done is a sudo dpi-update, nothing else. It goes from version 5.10.63+ to 5.10.81+.

popcornmix, 2021-11-25:
Yes, everything you have described is as expected.

You have moved to a new test kernel and you want to build (I assume) a kernel module which needs corresponding kernel headers.

rpi-source is a way of getting that. Or you could stick with the apt kernel if you don't want that inconvenience.

AzagraMac, 2021-11-25:
I need the kernel sources, I think I'll stay on 5.10.63+, which does have them.

popcornmix, 2021-11-25:
Kernel sources are available: https://github.com/raspberrypi/linux/

But you are free to stay on 5.10.63+.

closed 2021-11-25 by popcornmix

AzagraMac, 2021-11-25:
If with the command to update the firmware `rpi-update` I install the latest one, but it does not have the sources... it does not help me at all.

popcornmix, 2021-11-25:
And it's not meant to.

paulwratt, 2022-01-21:
For anyone ending up here:

> The headers from apt will match versions of kernel supplied through apt.

`sudo apt-get install raspberrypi-kernel-headers`

NOTE:
Be careful if you have `armhf` installed, but `arm_64bit=1` kernel enabled in your `/boot/config.txt`. The `..-v8+` are not in `armhf`, but should be in `aarch64` though (which wont have `..-v7` or `..-v7l`).