Hexxeh/rpi-firmware #275: rpi-updating on Raspbian Bullseye 32 bit (platform RPI4b), 64 bit kernel is installed

qtmrcdmc, 2023-03-17:
Executing rpi-update on Raspbian Bullseye 32 bit using latest commit ( rpi-update cba2ae16185afbb4895673b6c2ef7d4078abd65f ), 64 bit kernel (6.1.19-v8+) is installed instead of 32 bit one.

pelwell, 2023-03-17:
Both 32-bit and 64-bit kernels are installed. However, on a Pi 4 the default kernel is now the 64-bit one.

Is there a specific reason why you need the 32-bit kernel? You can force the firmware to load it by including `arm_64bit=0` in config.txt.

We believe the 64-bit kernel should be the best for most people. We are anticipating a few problems building software packages that assume the kernel architecture matches the userland architecture, but otherwise things should continue to work as (or better than) before.

popcornmix, 2023-03-17:
See: https://forums.raspberrypi.com/viewtopic.php?p=2088935#p2088935

qtmrcdmc, 2023-03-17:
Thank you. Right now we are using Raspbian 32 bit just because we started building sw on that distro, and we match kernel architecture with userland architecture, both 32bits. Woud it be safe continue using this approach?
Do you think that 32 bit distro would be still supported just as the 64 bit one in the next future?

Please, an argument that does not directly concern this aspect: we are looking for graphical improvements; for instance we notice that rescaling rendering chromium window (on xorg) to FHD (1920x1080), reduces global fps rendered.

In your opinion is there still room for improvement (on RAspberry Pi 4 b) in the use of VideoCore VI from this point of view?
Would it be necessary move to Wayland instead of Xorg?

XECDesign, 2023-03-17:
Heads up that you can preface commands with `setarch linux32` to get around with issues of build systems using uname to determine the architecture. For example, instead of running `./configure; make`, you'd run `setarch linux32 ./configure; setarch linux32 make`

pelwell, 2023-03-18:
By the way, the rpi-update and rpi-firmware repos are now officially hosted in our GitHub organisation: https://github.com/raspberrypi/rpi-update and https://github.com/raspberrypi/rpi-firmware.